exothink/eXoCAN #3: CAN2

mackelec, 2020-10-16:
Would you have any interest in implementing CAN2 in your library for the stm32F105/107?

  I have managed to get CAN2 working (as a stand-alone functions) , working along side your library on a stm32F105.  I told Arduino IDE that the board was a blue pill and it was happy.  (It had frequent upload timeouts - I think because on the 105 & 107 you can use any port for uploading so may take longer to scan ?)  My code is only basic and doesn't take advantage of all the features your library does.

You are welcome to my code but it is mostly based on your work.

damienmaguire, 2020-11-05:
I too would love to get this working on a F105. Please let me know where I CAN:) make a donation. Thanks for the hard work.

mackelec, 2020-11-07:
In case it is of any help, this is the board I used to test the Dual CAN on the STM32F105
[https://github.com/mackelec/STM32F105_Dual_CAN_Board](url)

exothink, 2021-02-04:
Currently, I have no plans to add more devices. Too many other projects. 🙁

closed 2021-02-04 by exothink