exothink/eXoCAN #6: eXoCAN

saadat925, 2021-04-16:
Hi 
I want to send 8 byte fixed data to id=0x0 and read 8 byte variable data from id=0x2bd .
Please guide me . Thanks
#

exothink, 2021-04-16:
in examples/bpCanBlink.ino

Find:
  can.begin(STD_ID_LEN, BR250K, PORTA_11_12_WIRE_PULLUP); // 11b IDs, 250k bit rate, no transceiver chip, portA pins 11,12
edit: for your hardware/baud rate

Find:
  can.filterMask16Init(0, 0, 0x7ff, 0, 0);                // filter bank 0, filter 0: don't pass any, flt 1: pass all msgs
replace with:
  can.filterList16Init(0, 0x2bd); 

Find:
  int txMsgID = 0x069;  set to 0x0   // I don't believe 0x0 is a valid CAN address though

Everything you need is in the examples.

closed 2021-04-16 by exothink

saadat925, 2021-04-16:
In the device manual, say send 01 to identifier 0.

saadat925, 2021-04-16:
Can the value of txDly be reduced to 100ms?

saadat925, 2021-04-16:
There is nothing in this program about the receive function for example 8 byte data.  It is only a comparison. Please also explain the receive function.

saadat925, 2021-04-17:
Dear sir
please answer my questions.  Thank