exothink/eXoCAN #12: Compiling on PlatformIO on Linux

delboy711, 2021-11-29:
Hi.  Thanks for the library it is just what I needed for my project. There are so few examples of using canbus on STM32 out there. This library is invaluable.

I did find the library would not compile using PlatformIO on Linux because of  line 25 in eXoCAN.h  
`#include <arduino.h>`

The 'A' needs to be capitalised
`#include <Arduino.h>`

The other thing I noticed is that the timeout in the function eXoCAN::transmit() is commented out. Is there a reason for this?
With the transmit FIFO being only three messages long in the STM32F103C8 it is useful to wait for a space in the FIFO when sending a cluster of messages.

exothink, 2021-12-01:
Interesting  the cap 'A' isn't required in Windows.  I remember now, Windows file manager doesn't distinguish.  I'll it update later.

stinkydiver73, 2022-10-10:
There is a pull request older than this issue :-)

closed 2023-10-18 by exothink