exothink/eXoCAN #13: send more than 1 CAN message in a row

epikao, 2021-11-30:
Hello

Is it possible to send more than 1 Can message in a row (without delay) and without data being lost at the recipient? How can this be implemented properly?

```
void canISR() // get CAN bus frame passed by a filter into fifo0
{
    can.receive(id, fltIdx, rxbytes);  // empties fifo0 so that another another rx interrupt can take place
}

void setup()
{
  can.attachInterrupt(canISR);
}

void loop()
{
...
...
    can.transmit(frame1...);
    can.transmit(frame2...);
    can.transmit(frame3...);
    can.transmit(frame4...);
    can.transmit(frame5...);
...
}

```

Thank you

epikao, 2021-12-01:
or perhaps I should ask differently?
As a sender - how do I know if the recipient has emptied the mailbox?
My suggestion: 
Sender sends, if the recipient empties the mailbox, the recipient also sends something, and if the sender receives the message from the recipient, the sender can send something again....

exothink, 2021-12-01:
Are you experiencing dropped messages or thinking they may get dropped.  The 103 handles them in hardware.  The FIFO can contain more that one RX message.  I have several on a bus sending and one receiving and it never experienced any drops.

I suggest you 'loop back' and compare sent and received data.  One sends the other receives it and echoes it back.  This first compares the two.

BTW there is a bit that indicates a FIFO over run.  It's all in the documentation.  

What data rate.

epikao, 2021-12-01:
> Are you experiencing dropped messages or thinking they may get dropped

Just thinking, and because I found some issues about in the internet, see for example here:
https://www.mikrocontroller.net/topic/370955 (but is another library)

But yes, thank you for your input, I will try it first with echoes. And with documentation I think you mean the STM32F103 datasheet / CAN part / register?...

BTW: For me it makes no sense why SN65HVD230 should not work. Do you have an idea why it was not working with your setup?  I have both chips (TJA and SN), and I will of course test them as soon as I can....

epikao, 2021-12-02:
I'm currently trying to get the interrupt blink example to work. Unfortunately without success (with TJA IC).
The only change I made in the Blink example is the change from PORTA_11_12_WIRE_PULLUP to PORTA_11_12_XCVR.

Is this correct? Or does it need more change with transceiver IC's ?

Thanks

epikao, 2021-12-02:
I tried now your example exoCanDev,

I always receive:

[image]

epikao, 2021-12-08:
Hello,

I just tested it (with your blinkInterrupt example), see code below. With no delay it does not work (it shows always only the first frame), with delay(5) between every transmit it works.... but I'm not happy with such a solution.

```
void loop()
{
  //if (millis() / txDly != last)                 // send every txDly, rx handled by the interrupt
  //{
    //last = millis() / txDly;
    //txData[1]++;
    can.transmit(txMsgID1, txData1, txDataLen);
    delay(5);
    can.transmit(txMsgID2, txData2, txDataLen);
    delay(5);
    can.transmit(txMsgID3, txData3, txDataLen);
    delay(5);
    can.transmit(txMsgID4, txData4, txDataLen);
    Serial.println("loop and transmit");
    delay(200);
  //}
}
```

exothink, 2021-12-09:
I believe you are incorrect.  I've looked at it on an oscilloscope and have seen one message sent immediately after the other without any issues.  You are sending print statements via serial IO.  It is the print statements that slow things down.

delboy711, 2022-01-04:
I am experiencing the same issue. 
If two messages are sent one after the other, only the first arrives.

The cause I found was that the eXoCAN::transmit() function returns false if the transmit mailbox 0 is not empty. 
When the first message is sent the function returns as soon as the message is queued but before the state machine has successfully sent it.
When the second message is sent the MBE  (mail box empty)  bit has not been set yet and the function returns without doing anything.

The solution was to pause waiting for the MBE bit.  This issue has clearly been seen before since there is already a suitable pause in the code which has been commented out. All I had to do was restore the pause.

`bool eXoCAN::transmit(int txId, const void *ptr, unsigned int len)
{
    
    uint32_t timeout = 10UL, startT = millis();
    while (periphBit(tsr, 26) == 0) // tx not ready
    {
        if((millis() - startT) > timeout) {
            return false;           //Timeout
        }
    }
    
    // TME0
    //if (periphBit(tsr, 26) == 0) // tx mailbox 0 not ready)
    //    return false;

    if (_extIDs)
        MMIO32(ti0r) = (txId << 3)  + 0b100; // // set 29b extended ID.
    else
        MMIO32(ti0r) = (txId << 21) + 0b000; //12b std id

    MMIO32(tdt0r) = (len << 0);
    // this assumes that misaligned word access works
    MMIO32(tdl0r) = ((const uint32_t *)ptr)[0];
    MMIO32(tdh0r) = ((const uint32_t *)ptr)[1];

    periphBit(ti0r, 0) = 1; // tx request
    return true;
}`

A better solution would be to use all three transmit mailboxes in sequence.

delboy711, 2022-01-05:
Here is the version of eXoCAN::transmit() I am using.
This one uses all three mailboxes so up to 3 messages may be sent consecutively. To send more ,the user must put a short delay between groups.  I tried using a while loop to wait until a mailbox became free, but that did not work for some reason.
```
bool eXoCAN::transmit(int txId, const void ptr, unsigned int len)
{
    uint8_t mbx = 0x00;                     //mailbox offset 0=mailbox 0, 0x10=mailbox1, 0x20 = mailbox 2
    bool mbx_empty = false;
    uint8_t i;
    for (i = 0; i<3; i++) {                      //Check each mailbox in turn to find an empty one
        if (periphBit(tsr, 26 + i) == 1) {
            mbx_empty = true;
            mbx=  i *  0x10;
            break;                                   //Mailbox empty
        }
    }
    if(!mbx_empty) return false; //No mailbox available try later

    if (_extIDs)
        MMIO32(ti0r + mbx) = (txId << 3)  + 0b100;       // // set 29b extended ID.
    else
        MMIO32(ti0r + mbx) = (txId << 21) + 0b000;     //12b std id

    MMIO32(tdt0r + mbx) = (len << 0);
    // this assumes that misaligned word access works
    MMIO32(tdl0r + mbx) = ((const uint32_t *)ptr)[0];
    MMIO32(tdh0r + mbx) = ((const uint32_t *)ptr)[1];

    periphBit(ti0r + mbx, 0) = 1; // tx request
    return true;
}
```

exothink, 2022-01-07:
Hello delboy711.   Thank you for posting your solution.  

In my examples I use the following to 'pace' the TX :  
       if (millis() / txDly != last)                 // send every txDly

This leaves it up to the application to make sure that they are not trying to send faster than the hardware is capable of.  Since the data rate and message length are variable in the application the minimum time between consecutive transmissions is also variable.  My design decision was maximize throughput by minimizing code overhead.  The 'check' that I commented out helps accomplish that and also keeps the code from blocking.  Blocking occurs if another device corrupts the bus, the 'check' stalls and the code freezes.

I also chose not to use the two additional mailboxes because once they are filled you have the same exact issue. 

My application has a dozen stm32f103s on the bus.  Some update data such as fuel level in tens of seconds, while tachometer data is updated in tens of milliseconds.

Regards,
John

closed 2022-01-07 by exothink